terraref/drone-pipeline #15: Fix issue where specifying job parameters file causes RabbitMQ to time out

Chris-Schnaufer, 2019-03-04:


Chris-Schnaufer, 2019-03-04:
This issue appears to be tied to the extractors-opendronemap.txt file having a setting set in it (not an empty file and not having everything commented out).

Chris-Schnaufer, 2019-03-18:
The problem is that there are settings that can cause ODM (Open Drone Map) to timeout the RabbitMQ communications.
Solutions tried:
* a separate thread runs ODM
* different initial ODM settings
* increasing CPUs, Memory, and disk space

Running the ODM software as a completely separate process has **not** been tried. To do this the following steps should take place:
1) Write an updated settings.yaml file for each run
2) Call the ODM software as a separate process
3) Wait for completion

dlebauer, 2019-04-08:
underestimated 1 should have been 3-5. had to take code and put it in a separate application

closed 2019-04-08 by KristinaRiemer