terraref/drone-pipeline #29: Document what data's needed to put plot boundaries into BetyDB, and the SQL calls to populate the DB

Chris-Schnaufer, 2019-03-26:
More information is needed on how to put plot boundaries from a shapefile into BetyDB for the drone pipeline

kimberlyh66, 2019-03-26:
I think you will need to use shp2pgsql to get postgis geometry from shapefile (unless you know of another method).

kimberlyh66, 2019-03-26:
To describe the experiment associated with a site, you will need to provide the experiment name. Start date, end date, design, and description are optional. The experiment can be associated with treatments. Minimum required information to describe a treatment is the treatment name. 

To describe the site, you will need to provide the site name and geometry. Other fields such as city, state, country, and time zone are optional.

Sites should be associated with a citation. You will need to provide the author, title, and year.

dlebauer, 2019-04-08:
should be a readme w/ code that demonstrates how to do this on a test instance of the database

put instructions for running test instances of database into our wiki

not sure where the rest of the instructions should be documented? but in the same place as the meta-data upload tutorials, so in tutorials/metadata-upload for now is fine.

dlebauer, 2019-04-08:
Could go in the BETYdb data entry documentation https://pecan.gitbook.io/betydbdoc-dataentry but need to check with Scott Rhode about the status of migration to bookdown. (ask on the pecan bety slack channel)

kimberlyh66, 2019-05-21:
@Chris-Schnaufer see pull request: terraref/tutorials#145

Chris-Schnaufer, 2019-06-17:
Chris needs to create ticket to get work done and then do the work

closed 2019-07-01 by Chris-Schnaufer