terraref/drone-pipeline #37: TERRA REF expect coordinates (geo-images, et. al.) to use EPSG 4326

Chris-Schnaufer, 2019-05-02:
The problem with TERRA REF's expectation is that source images, shapes, and other data may not be in the expected coordinate system (EPSG 4326). This may be due to error introduced by reprojecting data.

- [ ] Investigate the level of effort involved in changing TERRA REF to use an image's/data's coordinate system
- [ ] document the changes needed for above step
- [ ] determine plan for changing TERRA REF
- [ ] document what's expected for source data's coordinate system

Chris-Schnaufer, 2019-05-03:
- [ ] terrautils/betydb.py:
   - get_sites() assumes `kwargs['containing'].split(",")` coordinates are epsg:4326
   - get_sites_by_latlon() assumes epsg:4326 [calls above get_sites()]
- [ ] terrautils/extractors.py:
   - _get_bounding_box_with_formula(): hard coded location and translation, epsg:4326 [called from spatial.py]
- [ ] terrautils/images.py:
   - image_get_geobounds: assumes a known coordinate system (no transforms made on file coordinates) [recommend an SRID parameter?]
   - polygon_to_tuples: assumes a known coordinate system (no transforms made to polygon). Fixed: Use polygon_to_tuples_transform() if concerned about coordinate systems
- [ ] terrautils/formats.py:
   - create_geotiff: hard coded transform and epsg:4326. [use gdal to load transform and add optional SRID parameter?]
- [ ] terrautils/geostreams.py:
   - for all methods: check that an SRID is specified on GEO JSON strings
   - create_datapoints() and create_datapoint_with_dependencies(): not clear that SRID is part of location information [need to look into more/ask Max]
   - get_sensors_by_circle(): assumes a coordinate system [needs an SRID?]

Chris-Schnaufer, 2019-05-16:
Look at documentation and tickets regarding this

rickw-ward, 2019-06-06:
I am curious- are projections using epsg:4326 and  epsg:32612 (UTM 12N) identical?  Locally, I always use UTM 12N.

Chris-Schnaufer, 2019-07-22:
@rickw-ward Sorry for the delay in my response. I am not an expert in coordinate systems so I can't answer knowledgeably on this, but I do know there must be a difference in representation since they have different epsg numbers. Also, all conversion introduce errors - how serious the errors are is dependent upon the type of coordinate systems involved (rectilinear, spherical, other) and the desired level of accuracy.

Chris-Schnaufer, 2019-07-22:
The analysis is completed. Next step is to review with TERRA REF developers

Chris-Schnaufer, 2019-07-30:
Review Thursday meeting w/ IL Urbana

max-zilla, 2019-08-01:
@robkooper fyi

max-zilla, 2019-08-01:
Next steps: I'll take a close look at this list but overall I think these are good changes and we can start implementing on a branch whenever ready. David also recommends adding tests where possible.

closed 2020-02-05 by Chris-Schnaufer