terraref/drone-pipeline #42: Review documentation for submission of pipeline jobs

Chris-Schnaufer, 2019-06-20:
**Task to do**
Review getting into Clowder, adding files, submitting job using Jupyter notebook

Defines expectations
Defines processes

**Reason**
Comment on and improve the documentation and process

**Result**
Improved documentation for processing steps

**Steps to take**
- [ ] Obtain data
- [ ] Input into Clowder
- [ ] Start processing to generate table
- [ ] Validate results

cmbrown89, 2019-06-27:
This tutorial was written for the dockerized jupyter data science notebook and so there should be a tutorial on how to get that up and running since running a dockerized jupyter notebook is not an intuitive process:
- In that tutorial, it would be great to have some screenshots that accompany the steps
- That tutorial should probably give a brief overview of Clowder, the drone pipeline, the significance of running things in docker containers, what the jupyter data science notebook is, and provide some links for more info on all of those things.
- At the bottom, there could be a small Tips section where if they run into the same problems I did, they are provided with some things to try like needing to run `jupyter notebook list` to get around tokens/passwords and changing http://localhost:8888/tree to http://localhost:8888/lab for the cool GUI.

I went through the tutorial using Clowder 1.5.2 and I suggest the following changes:
- In the What To Expect section, a 1-2 sentence broad overview of the entire process would be nice just reiterating points for people who are new to this
- In Step 2, "focussed" should be "focused"
- In Step 2, I suggest adding a print(experiment) statement at the bottom to make sure user checks that they get what's below
- There's a warning that shows when the last step is run in the pipeline making you think that there's been an error
- There was some weirdness when I tried seeing whether any output folders had been created from the pipeline in My Spaces. After refreshing a few times, the folders popped up in My Spaces.

Chris-Schnaufer, 2019-06-27:
Created a new ticket re: using Jupyter docker container.

The last bullet under suggested changes from above comment regarding Clowder not showing the new space created: no action taken since this is a Clowder issue and doesn't impact the tutorial

dlebauer, 2019-07-11:
review requirements w/ @Chris-Schnaufer and then close or move to next milestone!

ALso please link to the documentation in question

Chris-Schnaufer, 2019-07-25:
Closing as done. https://github.com/terraref/drone-pipeline/blob/develop/tutorials/submit.ipynb

closed 2019-07-25 by Chris-Schnaufer