terraref/drone-pipeline #51: Define indices that can be calculated from RGB images

dlebauer, 2019-07-11:
And if you come across them then also please document methods for multispectral

Identify 3-5 indices that are different. Or catalog them all as you go. 

See also ridb https://github.com/ropenscilabs/ridb

And https://terraref.ncsa.illinois.edu/bety/variables?utf8=%E2%9C%93&DataTables_Table_0_length=25&search=index&direction=&sort=

dlebauer, 2019-08-01:
@cmbrown89 could you link to the google doc here?

cmbrown89, 2019-08-01:
I've linked it above (Google Doc with Indices). Please let me know if the link works for you.

dlebauer, 2019-08-05:
@cmbrown89 I would propose starting with the following:
* Excess greeness index = 2g-(r+b)
* percent green = g/(r+b+g)
* Vegetative index = g/(r^a*b^(1-a)); a = 0.667

I think this issue can be closed and another one opened to cover adding these indices.

Once you get these in, it would be worthwhile to add other indices and combined indices that you list in this document.

KristinaRiemer, 2019-08-06:
@cmbrown89 I have a couple general questions about this. Are you gathering these indices to use and/or compare on a certain dataset? And these are only for RGB images, right?

cmbrown89, 2019-08-06:
Thanks @dlebauer I've started with those indices. I'll write the ticket/add relevant links later today.

@KristinaRiemer That's correct, these are only for RGB images. As for your other question, my understanding is that people would need to evaluate what their datasets called for since some indices will be more appropriate for certain experimental designs/plant types/leaf colors, ect. But my intuitive (and uninformed) theory is that there's probably a core set of RGB indices that can be generally used for most datasets. So I want to focus on those. Was that helpful enough?

dlebauer, 2019-08-13:
There is a new issue to follow up on this

closed 2019-08-13 by dlebauer