terraref/drone-pipeline #54: Drone Pipeline TravisCI integration: find quick solution to image size issue from ODM

Chris-Schnaufer, 2019-07-30:


Chris-Schnaufer, 2019-08-06:
@dlebauer Here are what i've found out.
Background: I am trying to use TravisCI to validate OpenDroneMap extractor ortho mosaic (OM) image generation. The OM images from run-to-run vary widely preventing any meaningful comparisons - thereby we're unable to validate the output with the generated images as they are. The main approach is to subtract a newly generated image from a master image and analyze the differences via a per-channel (R,G,B) histogram of pixel differences

1. By clipping out a geographic region from the generated ortho mosaics we get better results
2. The geographic clipping may have dimensional variations of 1 pixel (for my sample sets)
3. By moving around the area of comparison when sizes are different, I can get better results
4. From the image that results from the subtraction, we have less than 3% total pixels in any histogram bucket
5. The bad news is that from histogram position 13 to 255 (representing channel pixel value differences from slightly different to completely different) we have a cumulative difference for about 74% of the pixels

Looking at the original, clipped, RGB images it's easy to see why there's this problem - the generated images have differences that are noticeable.

I propose we document and compensate for the differences for our acceptance test.

Chris-Schnaufer, 2019-08-08:
Per conversation with David, calculating a % greenness value over the entire extracted (by geo-location) image with a margin of error  would be acceptable.

dlebauer, 2019-08-13:
half percent or less error is way lower than biological variation