terraref/drone-pipeline #55: Review findings with TERRA REF group

Chris-Schnaufer, 2019-07-30:


KristinaRiemer, 2019-08-08:
@Chris-Schnaufer should schedule meeting with Max to discuss these.

Chris-Schnaufer, 2019-08-16:
Meeting notes:
- BETYdb.py
    - Add EPSG parameter
- extractors.py:
    - Remove from extractor.py
- imagefile.py:
    - Add SRID parameter
- Formats:
    - Add EPSG parameter w/ default
- geosgtreams.py
    - There’s a newer geostreams library available
    - Max looks into new geostreams
    - Max looks into Clowder code for EPSG requirements
    - Document TERRA REF EPSG 4382 requirement
    - If used outside TERAA REF we need to make code changes
- Lemnatec.py
    - Timezone needs to be documented
    - Timezone is a known issue with BETYdb
- Spatial:
    - cliop_las and clip_raster add EPSG parameter
    - utm_to_latlon, scanalyzer_to_utm,. tuples_to_geokson, tuples_to_utm, _get_bounding_box_with_formula into its own file
    - otherwise document as suggested and add EPSG parameter as suggested
- Update TERRA REF README about coordinate systems

Chris-Schnaufer, 2019-08-16:
Added ticket #63 as the next step