terraref/drone-pipeline #72: I would like to know if the UAS User Log would be useful for collecting drone operations metadata

dlebauer, 2019-08-29:
We need a way of collecting metadata from the field. 

The UAS User Log collects metadata from a drone operator, and was recently released on GitHub https://github.com/DigitalAgricultureDiscovery/uas-user-log

It walks users through pages to learn about the flying conditions, platform, and sensors. And generates json metadata.

Would this be useful, either as an app or as a template for learning what data is useful to collect, and how to store it?

Chris-Schnaufer, 2019-09-26:
Below is the link to a copy of the generated JSON containing dummy data (extension renamed for GitHub support - original extension is '.json').

Looking at the produced data the following are available (notes follow):
- crop_Variety: could be mapped to BETYdb cultivar.name or BRAPI germplasmName
- general_FlightDate: use as the date for a timestamp
- general_Flights.Start: use as the time for a timestamp
- general_Flights.Latitude and general_Flights.Longitude: can be used to determine GMT offset of timestamp
**Note**: the general_Flights key contains an array of flights. If more than one flight is defined a decision needs to be made as to which to use (perhaps the earliest one that corresponds to EXIF data in images to be processed?)
**Note**: timestamps correspond to BRAPI observationTimeStamp

Missing from this JSON are: season, studyName, and collectingSite as defined by BRAPI. These BRAPI fields currently translate to season, experiment, and sitename for the Drone Processing Pipeline.

File generated at https://www.uasuserlog.org/
[https://app.zenhub.com/files/160253698/03fbf242-00de-4f34-9181-9c41fe51a80a/download](https://app.zenhub.com/files/160253698/03fbf242-00de-4f34-9181-9c41fe51a80a/download)

dlebauer, 2019-09-26:
What are your thoughts on how difficult it would be to build on what they have vs starting from scratch for collecting this information? Is most of the information useful? Does this overlap with the ESIP / LANDRS api project?

Chris-Schnaufer, 2019-10-07:
Collaborate with them to add missing fields: create an issue on their site and offer to do it

closed 2019-10-07 by Chris-Schnaufer