terraref/drone-pipeline #82: Create alternate means of passing plot name forward

Chris-Schnaufer, 2019-09-24:
**Task to do**
The current method of passing along plot names is to use Dataset naming conventions. Adding the plot name to metadata is a better way of making the name available

**Reason**
The current way of using a name to hold the plot name is problematic and can cause problems. By moving to a metadata based approach the plot name is isolated and can be accurately determined

**Result**
A consistent and stable way of specifying plot name helps stabilize extractors

dlebauer, 2019-11-18:
How does this integrate with plot names and boundaries in BETYdb? Once plot boundaries are specified, they should be kept through the season by default, though in some cases (e.g. if some plants do not germinate) the defined plot boundaries will not be the same as those used for plot clipping. 

One piece of QA/QC in a report that would be nice to have is a low res full field map with plot polygons and polygons used to clip overlaid (and perhaps named or numbered?)

Chris-Schnaufer, 2020-01-31:
Closing as no longer relevant given Makeflow based pipeline

closed 2020-01-31 by Chris-Schnaufer