terraref/drone-pipeline #86: Add ability to handle LINESTRING, POLYGON, POLYGON Z to MULTIPOLYGON in plotclip extractor

Chris-Schnaufer, 2019-09-27:


KristinaRiemer, 2020-02-13:
Might need to discuss this with @dlebauer

dlebauer, 2020-02-13:
I'd rather put constraints on what we will accept and how to handle exceptions

- Polygon and Polygon Z should both be coerced to polygon and used
- Multipolygon 
   * if 1 polygon coerce to polygon
   * if > 1 polygon, throw an error and track down original intent 
- Linestring:
   * if it is a closed polygon, coerce to polygon (could also just coerce to polygon w/ convex hull)   
   * if it doesn't close, throw error

Chris-Schnaufer, 2020-02-13:
This sounds good. Can I assume that additionally:
1. multi-linestring has the same constraints as multi-polygon
2. if not a polygon type or closed-linestring type we throw an error (eg: point, multi-point, multi-geometry)

dlebauer, 2020-02-17:
Yes
Yes

-- 
David LeBauer
Director of Data Sciences
CALS Comm and Technologies
THE UNIVERSITY OF ARIZONA

BSRL, 207
1230 N Cherry Ave | Tucson, AZ 85721
Office: 520-621-4381 | Cell: 760-468-8621
dlebauer@arizona.edu

*datascience.cals.arizona.edu <https://datascience.cals.arizona.edu>*
orcid <https://orcid.org/0000-0001-7228-053X> | twitter
<https://twitter.com/dlebauer> | github <https://www.github.com/dlebauer> |
linkedin <https://www.linkedin.com/in/dlebauer/>

Chris-Schnaufer, 2020-04-09:
Needed for Francilino's data

Chris-Schnaufer, 2020-04-27:
Using envelope calculation to ignore shape types

closed 2020-04-27 by Chris-Schnaufer

Chris-Schnaufer, 2020-04-27:
https://github.com/AgPipeline/transformer-plotclip/pull/7