terraref/drone-pipeline #98: Fork TERRA REF repos as needed to new GitHub organization

Chris-Schnaufer, 2019-10-18:


dlebauer, 2019-10-18:
clean up as you go

Chris-Schnaufer, 2019-10-31:
To keep history:
1. Fork the TERRA REF repo to another organization (where you have sufficient permissions)
2. Rename the new fork to something meaningful using correct naming conventions
3. Perform merges into `master`, delete unneeded branches, and do other branch cleanup tasks
4. Checkout the repo to a work area
5. `git` move, delete, etc. until the wanted files are where you need them to be and everything else is removed
6. Pull the template code to the work area (if applicable). You will most likely need the '--allow-unrelated-histories' flag
7. Develop your transformer cleaning up as you go, and checking in changes
8. When satisfied, perform a `git push --all` to the final repo

Something not noted is that you will probably want to use `git remote` commands to add and remove git endpoints as needed. For example:
- git init && git remote add origin <url to original>
- git fetch && git checkout master && git pull origin master
- git add remote template <template url> && git fetch template
- git pull --allow-unrelated-histories template master
- git remote rm template
- Much deleting and moving happens here
- Much code development and testing happens here
- git commit -m "Lots of changes" && git push
- git remote rm origin
- git remote add origin <final destination url>
- git push all

closed 2019-11-04 by Chris-Schnaufer